AlexGyver/GyverLibs #1: Несколько замечаний

tamtakoe, 2018-07-05:
Отличная библиотека! Стоит развивать дальше.

1. Переменную mainTimer стоит переименовать в gMillis (g - Gyver или Global) или, в крайнем случае, в mainTimerMs. Суть в том, что имя переменной должно подсказывать в каком формате хранятся в ней данные.

2. Если buf это время до ближайшей задачи, то можно сделать его типом int. Ну и переименовать в что-то более понятное. Например nextTaskMillis или nextTaskTimeMs

3. Насколько понимаю, расширенный режим нужен лишь для настройки энергосбережения. 
```
myRTOS.tick(&buf);
LowPower.powerDown(SLEEP_PERIOD, ADC_OFF, BOD_OFF);
myRTOS.wake(&buf, &mainTimer);
```
Думаю, лучше использовать коллбэк, в котором пользователь реализует нужный ему режим
```
myRTOS.tickAndSleep(&buf, &mainTimer, powerModeSetter);

// Если в ардуиновском Си есть перегрузка или опциональные аргументы:
myRTOS.tickAndSleep(&buf, &mainTimer, SLEEP_PERIOD, powerModeSetter);
myRTOS.tickAndSleep(&buf, &mainTimer, SLEEP_PERIOD, ADC_OFF, BOD_OFF);

// Если нет:
myRTOS.tickAndSleepLowPower(&buf, &mainTimer, SLEEP_PERIOD, ADC_OFF, BOD_OFF);
myRTOS.tickAndSleepCustom(&buf, &mainTimer, SLEEP_PERIOD, powerModeSetter);
```
4. Опять же не очень понятно, зачем каждый раз передавать &buf, &mainTimer в GRTOS и GTask. Логично сохранить эти указатели один раз при инициализации. Тогда код ещё упростится.
```
myRTOS.tickAndSleep()
// или просто
myRTOS.tick()
```

5. buf и mainTimer можно сделать опциональными и сохранять по умолчанию в регистр общего назначения (они есть почти у любого МК)

6. Подумать по поводу вызова задачи по прерыванию, чтобы не было сбоев из-за этого

7. Прикрепление задач к общей задаче, а потом вкладывание её в лист кажется несколько избыточным. Могло бы быть как-то так
```
#include "GyverRTOS.h"

GRTOS myRTOS;
Gtask task3;

// Если setPeriod не задан, или у какой либо задачи таймаут будет меньше
// минимального WatchDog периода, то режим энергосбережения не применяется.
// Задачи, добавленные без параметров, выполняются с нулевой задержкой по кругу 
// как в многозадачной ОС.
void setup() {
  myRTOS.setPeriod(SLEEP_15MS);
  myRTOS.addTask(action1);
  myRTOS.addTask(action2); 
  myRTOS.addTask(action3, 1000);
}

void action1() {
  Serial.println("action1!");
}
void action2() {
  Serial.println("action2!");
}
void action3() {
  myRTOS.removeTask(action3);
}

void loop() {  
  myRTOS.tick();
}
```
В общем, если доработать, то будет очень полезной штукой

AlexGyver, 2018-09-05:
Спасибо за фидбек, библиотеку сделал просто ради научного интереса. Можете плиз исправить хотя бы треть вышеуказанного (главное 7 пункт) и подтянуть реквест? Я не знаю всех "фишек" С++, и некоторые вещи просто не понимаю как сделать =) Ради этой библиотеки вообще пришлось гуглить что такое указатель например. Спасибо!

tamtakoe, 2018-09-06:
Я тоже не специалист по C++, но попробую что-нибудь сделать)

closed 2019-12-14 by AlexGyver