AlexGyver/GyverLibs #5: AlexGyver

WonderCRM, 2018-12-27:
AlexGyver, доброго времени суток, спасибо за библиотеке.
Есть небольшое предложение по библиотеке GyverButton.
Предположим, требуется задать задержку (setTimeout ),
скажем в 30 сек, при этом срабатывание 2х, 3х и т.д. кликов происходит
с такой же задержкой, что дико не удобно.
Предлагаю параметр setTimeout, разделить на 2 (setTimeout и setHoldTimeout).
Первый это время задержки нажатой кнопки, а второй, 
время, после которого пачка кликов является завершённой.

З.Ы. для себя код подправил, но может кому надо будет, готовый вариант могу скинуть, для размещения.

AlexGyver, 2019-03-03:
Привет, я забыл где ты об этом писал. Сразу добавил новый метод, а куда говорить спасибо за идею потерял)

WonderCRM, 2019-03-04:
Спасибо, что внес изменения, пока не пробовал твой последний вариант, использую ранний?отредактированный.

Ещё может есть смысл исключить функции одиночный-тройной клили.
Ведь на проверку каждой требуется хоть и малое, но время. Может лучше оставить проверку нажатий, а уже затем сравнивать с количеством. Так при неиспользовании кнопки меньше телодвижений.

AlexGyver, 2019-03-04:
Код построен так, что если в нем нет функции опроса двойного клика, то никакое время не тратится. А городить ради двойного клика функцию подсчёта кликов ну такое

WonderCRM, 2019-03-04:
> А городить ради двойного клика функцию подсчёта кликов ну такое
Так она вроде и так есть. Ну да ладно, смотри сам, тебе виднее ;).

AlexGyver, 2019-03-04:
Функция подсчёта кликов нужна для большого количества кликов. А двойной и тройной часто используются в жизни

closed 2019-11-12 by AlexGyver