AlexGyver/GyverLibs #6: Button.isPress() отрабатывает при запуске

catinwarmhands, 2019-03-07:
Здравствуйте. У меня возникла проблема с использованием библиотеки GyverButton.

Дело в том, что при запуске ардуины почему-то отрабатывает одно нажатие на кнопку (isPress()).

Поскольку я новичок в этом деле, то не понимаю - это мой косяк или же ошибка в работе библиотеки.

Вот мои тесты:

## Тест 1

Использую код с сайта, без изменений:
```c++
/*
   Данный код демонстрирует переключение режимов работы при помощи кнопки
   Для удобства используется библиотека отработки нажатий кнопки
   В этом варианте примера функции "режимов" вызываются только один раз
*/
#define PIN 3        // кнопка подключена сюда (PIN --- КНОПКА --- GND)
#define MODE_AM 5    // количество режимов (от 0 до указанного)
#include "GyverButton.h"
// моя библиотека для более удобной работы с кнопкой
// скачать мождно здесь https://github.com/AlexGyver/GyverLibs
GButton butt1(PIN);  // создаём нашу "кнопку"
byte mode = 0;       // переменная режима
void setup() {
  Serial.begin(9600);
}
void loop() {
  butt1.tick();             // обязательная функция отработки. Должна постоянно опрашиваться
  if (butt1.isPress()) {    // правильная отработка нажатия с защитой от дребезга
    // увеличиваем переменную номера режма. Если вышла за количество режимов - обнуляем
    if (++mode >= MODE_AM) mode = 0;
    // всё переключение в итоге сводится к оператору switch
    // переключение и вызов происходит только при нажатии!!!
    switch (mode) {
      case 0: task_0();
        break;
      case 1: task_1();
        break;
      case 2: task_2();
        break;
      case 3: task_3();
        break;
      case 4: task_4();
        break;
    }
  }
}
// наши задачи, внутри функций понятное дело может быть всё что угодно
void task_0() {
  Serial.println("Task 0");
}
void task_1() {
  Serial.println("Task 1");
}
void task_2() {
  Serial.println("Task 2");
}
void task_3() {
  Serial.println("Task 3");
}
void task_4() {
  Serial.println("Task 4");
}
```

После старта в порт сразу приходит "Task 1", хотя на кнопку я ещё даже не нажимал.
[image]

[image]

## Тест 2

Пробую переключать светодиоды. при mode=0 зеленый, иначе красный

```c++
#include "GyverButton.h"

const int goodPin = 8;
const int badPin = 7;
const int buttonPin = 3;

GButton button(buttonPin);
int mode = 0;

void setup()
{
   pinMode(goodPin,OUTPUT);
   pinMode(badPin,OUTPUT);
} 

void loop()
{
  button.tick();
  if (button.isPress()) {
    mode++;
  }
  if (mode == 0)
  {
    digitalWrite(badPin,LOW);
    digitalWrite(goodPin,HIGH);
  }
  else
    digitalWrite(badPin,HIGH);
    digitalWrite(goodPin,LOW);
}
```

[image]

На видео видно, что на долю секунды зажигается зеленый светодиод, но потом отрабатывает блок  ` if (button.isPress())`, и режим меняется на 1 (красный светодиод).
[Видео работы](https://youtu.be/VdzZfni9EfI)

Так же я пробовал подключать больше светодиодов с цикличным переключением кнопкой, но результат такой же - после старта загорается не нулевой светодиод, а первый.

Прошу прощения за некрасивые схемы и сборку - я только недавно начал осваивать всё это дело.

WonderCRM, 2019-03-07:
Где вы взяли схему подключения?
В этой библиотеке используется 2 ноги для кнопки, PIN - GND либо PIN - 5V, записит от параметров в сетапе. По умолчанию стоит  PIN - GND.

Попробуйте для начала этот пример

```
/*
   Пример использования библиотеки GyverButton, 1- 2- 3- нажатие
*/

#define PIN 3				// кнопка подключена сюда (PIN --- КНОПКА --- GND)

#include "GyverButton.h"
GButton butt1(PIN);

void setup() {
  Serial.begin(9600);
}

void loop() {
  butt1.tick();  // обязательная функция отработки. Должна постоянно опрашиваться

  if (butt1.isSingle()) Serial.println("Single");     // проверка на один клик
  if (butt1.isDouble()) Serial.println("Double");     // проверка на двойной клик
  if (butt1.isTriple()) Serial.println("Triple");     // проверка на тройной клик
}
```

А так лучше режим авто опроса, пример тут
Gbutton_example_auto.ino

catinwarmhands, 2019-03-07:
@WonderCRM большое спасибо, разобрался.
Такая схема подключения была в туториале у китайцев в комплекте с ардуиной.

closed 2019-03-07 by catinwarmhands