AlexGyver/GyverLibs #9: GyverUART

PetrakovKirill, 2019-08-03:
1. Глобальные переменные торчат наружу, это может вызвать конфликты, или привести с побочному-использованию.

2. Реализация приема реализована крайне не плохо:
   - никак не проверяется граница буфера, это значит, что если пользователь по каким- 
     то причинам не вычитает буфер _UART_RX_BUFFER, а прием не прекратится, то счетчик 
     может выйдет за границы буфера и начнется перезапись чужих данных.
  - Реализация uartRead со смещением всего буфера - крайне не оптимальное решение. 

3. `	uint32_t integer = data;`
   `    uint32_t fract = ((float)data - integer) * 1000000000;`
тип uint32_t не может в себя вместить число больше чем 2^32 - 1, что выдаст не верный результат при достаточно больших значениях data. В оригинальный код arduino учитывает этот нюанс (хотя и довольно топорно, но все же). Реализация поддержки преобразования float в строку от Atmel достаточно весомая (не зря по умолчанию она отключена, и её нужно добавлять руками), но, тем не менее, делает это корректно.

AlexGyver, 2019-08-31:
Версия 1.1
1. Глобальные сделал более нечитаемыми
2. Буфер заменён на циклический, теперь read занимает 0.7 мкс
3. Тут да, но было принято решение полностью отказаться от строк. Для отладки пойдёт, а реальному проекту врядли нужно будет с таким работать...

PetrakovKirill, 2019-09-01:
volatile char _UART_RX_BUFFER[UART_RX_BUFFER_SIZE];
volatile uint8_t _UART_RX_BUFFER_HEAD;
volatile uint8_t _UART_RX_BUFFER_TAIL;

Эти переменные по прежнему глобальные. Так как в любом файле можно написать, например:
extern volatile uint8_t _UART_RX_BUFFER_HEAD;
и можно манипулировать, спокойно читать и переписывать их.
для того, что бы инкапсулировать данные, нужно объявить их как static, тогда эти переменные будут доступны только в пределах файла.
Можно упаковать всю работу с UART в класс, но тут проблема с прерываниями, которую в принципе тоже можно решить.

AlexGyver, 2019-09-01:
Да кому нужны переменные с таким названием? Тем более extern. 
Статиком сделаю, в класс специально не стал оборачивать

PetrakovKirill, 2019-09-01:
Дело не в том, нужны эти переменные кому, или не нужны, инкапсуляция данных - это хорошая практика, даже от самого себя. Глобальные переменные рушат декомпозицию, и открывают доступ к внутренним компонентам. которые потенциально могут привести к побочному использованию, что делает компонент не стабильным. Это все конечно может показаться философскими рассуждениями, но лучше сразу позаботиться об этом. Как говорится, даже не заряженное ружье хотя бы раз да выстреливает.

AlexGyver, 2019-09-01:
Ну вот в ядре переменные времени тоже не спрятаны, их можно экстернить, и это удобное)

PetrakovKirill, 2019-09-02:
Если это так удобно, зачем тогда нужна функция millis?)

AlexGyver, 2019-09-02:
Миллис не тупо возвращает переменную потому что, там расчет идёт на основе счётчика переполнения и текущего счётчика. Удобно что можно счётчик переполнений как раз сбросить в 0 в ходе программы

PetrakovKirill, 2019-09-02:
Как раз таки тупо возвращает (это твоя реализация):
unsigned long millis() {	
	cli(); // остановить счет
	unsigned long m = timer0_millis; // перехватить значение
	sei(); // продолжить счет
	return m; // вернуть миллисекунды
}
Но, есть нюанс: переменная эта меняется в обработчике прерывания, т.е. асинхронно.
проц 8-битный, и для того чтобы переписать 32-х разрядное число нужно несколько инструкций, т.е. операция не атомарная. millis реализует атомарное чтение переменной, за счет блокировки прерываний. Зачем нужна атомарность и какую проблему это решает:
Предположим, что у тебя есть код, который должен сработать на строго определенное значение времени, т.е. проверка на равенство, для простоты предположим проверяем на равенство с 0.
Наступил момент код начал сброс твоей глобальной переменной и вот посреди операции (проц не может за одну инструкцию это сделать, и скажем текущее состояние числа 0xABCD0000, т.е. 2 байта успели обнулиться) вызывается прерывание, и происходит инкремент (теперь число 0xABCD001). Программа продолжает свое выполнение, и в конце работы обнуления глобальной переменной, её значение будет не 0, а 1, и твое условие не выполнится).
Этот пример очень утрированный, но вполне реальный. Но может быть и другой исход, все зависит от того, что записывается и когда.

AlexGyver, 2019-09-02:
А точно, с микрос перепутал. 
По остальному не понял... Понял что если я руками полезу трогать переменную миллис, то можно всё вообще запороть, потому что будет частичная запись, верно?
Но это если я знаю название переменной и могу её extern сделать

PetrakovKirill, 2019-09-02:
Да, поэтому лучше прятать реализацию подальше от пользователя, потому что, кому- то может показаться это более быстрым, чем если функцию вызывать.
В программировании такое явление называется Data Race, или гонки данных. В интернете, скорее всего, примеры будут для многопоточных приложений на ПК, но работа с прерываниями  имеет те же проблемы. Такие баги как правило нельзя воссоздать однозначно, а значит трудно отлавливать и анализировать, поэтому лучше сразу на корню рубить все концы, которые могут вести к подобным проблемам.

closed 2019-11-12 by AlexGyver