AlexGyver/GyverLibs #13: TM1637 with 6-digits display from RobotDyn

Tomasina8, 2019-08-14:
Купил 6-разрядный дисплей RobotDyn (https://robotdyn.com/catalog/displays/6-digit-led-display-tube-7-segments-76x19mm-tm1637.html)
Родной библиотеки нет. Вообще, ни с одной из найденных библиотек не работает корректно - вместо 6 цифр выводятся только 4, причем в странном порядке: вместо "123456" выводится "321хх4" (х - это пустое место). Т.е. первый (слева) разряд выводится в 3-е знакоместо, второй разряд выводится во 2-е знакоместо, третий разряд выводится в 1-е знакоместо, 4-й разряд выводится в 6-е знакоместо, а 5й и 6й разряд не выводятся вообще. 
Схема на этот дисплей: [ссылка](https://robotdyn.com/pub/media/0G-00005561==Mod-LED-Display-6D-TM1637-76x19mm/DOCS/Schematic==0G-00005561==Mod-LED-Display-6D-TM1637-76x19mm.pdf)
По даташиту поддержка 6 разрядов есть, но ни в одной библиотеке не реализована. 
Так как библиотека GyverTM1637 - одна из самых удобных и функциональных, вопрос - будет ли в новых версиях поддержка этого дисплея?

Anymorf, 2021-01-31:
> Купил 6-разрядный дисплей RobotDyn (https://robotdyn.com/catalog/displays/6-digit-led-display-tube-7-segments-76x19mm-tm1637.html)
> Родной библиотеки нет. Вообще, ни с одной из найденных библиотек не [...]

Я могу конечно ошибаться. Но
Судя по распиновке, с картинки это дублированый 2*3 символа. поесть повторяет 2 раза по 3 символа.
Видны дорожки от Сегментов 1,2,3 и от них же идут на Оба индикатора

Но это догадки, в наличии не имею, проверить не могу

AlexGyver, 2021-03-15:
Идёт разработка

closed 2021-03-15 by AlexGyver