AlexGyver/GyverLibs #15: Добавить бы метод проверки состояния

atomoc, 2019-11-09:
Часто использую таймеры как функцию setTimeout в js, только в данном случае приходится в мэйн функции останавливать таймер методом .stop(), и в момент нужного события, например if (button.isSingle()) запускать timer.setTimeout(ms); timer.start(); и в if (timer.isReady()) опять останавливать таймер. Не сказал бы, что прям очень удобно, но удобней пока не было придумано, а у самого пока от идеи только недовольство) В целом такой метод меня устраивает, но хотелось бы его усовершенствовать добавлением метода timer.isEnabled(). Сделай, пожалуйста.

atomoc, 2019-11-09:
А вообще, моё недовольство легко бы решилось, если бы был убран метод start, а вместо него использовался setInterval и setTimeout. В методе  isReady тогда нужно сделать проверку на тип таймера (интервал или таймаут) и, если таймаут, запускать stop().

И ещё, непонятно, для чего reset()? Пробовал его использовать, но он вообще не работал, правда это было не знаю в какой версии, может ты и исправил его.

AlexGyver, 2019-11-12:
Если честно, не понял вообще, что как зачем надо добавить. Ресет работает и работал всегда, он сбрасывает счётчик периода, т.е. если таймер стоит на 10 секунд, и с момента последнего срабатывания прошло 5 секунд и ты делаешь ресет, то таймер сработает через 10 секунд в следующий раз, очевидно

atomoc, 2019-11-12:
Я по всякому пробовал проверять ресет, не буду углубляться в воспоминания, к тому же они зачастую меняются на физ уровне, может что то и упустил, а может сейчас вру.

Логику не надо будет пересматривать если оставить всё как есть.

Суть в том, что в js setTimeout и setInterval уже содержит в себе start, к этому все привыкли, почему бы здесь не сделать так же?

AlexGyver, 2019-11-12:
Я не знаю кто и к чему привык, потому не понимаю вообще о чём идёт речь...

atomoc, 2019-11-12:
будет не так: timer.setInterval(3000);... timer.start();.... timer.stop();... timer.start();
а так: int timerInterval = 3000;... timer.setInterval(timerInterval);... timer.stop();... timer.setInterval(timerInterval);

AlexGyver, 2019-11-12:
И зачем, в каких ситуациях это использовать?

atomoc, 2019-11-12:
И так: int timerInterval = 3000;... timer.setTimeout(timerInterval);... timer.stop();... timer.setInterval(timerInterval);

atomoc, 2019-11-12:
В тех же самых. Просто меньше кода из-за отсутствия .start()

AlexGyver, 2019-11-12:
Я так понял, что не хватает автоматического старта в setInterval? А в каких ситуациях нужно постоянно включать и выключать таймер?

atomoc, 2019-11-12:
Важней не это, а именно setTimeout, вот это чудо из чудес ;-)

atomoc, 2019-11-12:
Я уже третью программу без этого никак не могу обойтись

AlexGyver, 2019-11-12:
Можно в таком случае все понятно и подробно расписать, что и как должно работать? Потому что я не понимаю зачем это нужно и зачем так использовать таймер

atomoc, 2019-11-12:
[image]

atomoc, 2019-11-12:
Попробую

AlexGyver, 2019-11-12:
Таймаут - случаем не о ручном режиме таймера идёт речь? Здесь он называется так.

atomoc, 2019-11-12:
О ручном всмысле не использовать tick()? В таком случае разве вообще что то будет работать?

atomoc, 2019-11-12:
К тому же это непонятно для новичков. Проблем, согласен, добавление setTimeout может добавить в коде использующем старые версии, но это можно обойти установкой заглушки на метод start()

AlexGyver, 2019-11-12:
> О ручном всмысле не использовать tick()? В таком случае разве вообще что то будет работать?

Ручной режим работы таймера

atomoc, 2019-11-12:
setTimeout будет работать точно так же как и setInterval, пока время цикла не наступит. Как только время цикла наступит, сработает stop(), в чём и будет принципиальное отличие его от setInterval

AlexGyver, 2019-11-12:
> К тому же это непонятно для новичков. Проблем, согласен, добавление setTimeout может добавить в коде использующем старые версии, но это можно обойти установкой заглушки на метод start()

Если честно все ещё не понимаю ничего. Добавить старт в setInterval - возможно стоит это сделать, да. Убирать старт - глупость, лично я часто пользуюсь остановкой таймера, и запуском его на тот же период

AlexGyver, 2019-11-12:
> setTimeout будет работать точно так же как и setInterval, пока время цикла не наступит. Как только время цикла наступит, сработает stop(), в чём и будет принципиальное отличие его от setInterval

Это ручной режим таймера

atomoc, 2019-11-12:
Нашёл

atomoc, 2019-11-12:
Правда непонятно, я же видел твой пример и у меня почти получилось завести его в уме, прям тепло лампочки ощущал, но так и не смог понять

atomoc, 2019-11-12:
Спасибо за разъяснение, конечно

AlexGyver, 2019-11-12:
> setTimeout будет работать точно так же как и setInterval, пока время цикла не наступит. Как только время цикла наступит, сработает stop(), в чём и будет принципиальное отличие его от setInterval

При проверке на if .isReady выдаст тру только один раз, дальше будет возвращать фолс? Или до перезапуска тру?

atomoc, 2019-11-12:
Да, только один раз

AlexGyver, 2019-11-12:
Без проблем, давай допилим библиотеку под любителей js. Только мне нужно понять, что и как должно работать, и главное - зачем и где это использовать

atomoc, 2019-11-12:
Могу переделать твой пример Manual. Принцип то идентичен

AlexGyver, 2019-11-12:
Значит туду:
1. Добавить setInterval как синтаксический аналог setPeriod, чисто для привычности
2. setPeriod (setInterval) запускают таймер (также сбрасывая период, там внутри reset)
3. setTimeout делает тож самое, но автоматически пкреключает режим таймера на ручной
Все верно?

atomoc, 2019-11-12:
100%

AlexGyver, 2019-11-12:
Оки, посплю чутка и утром уже выкачу апдейт

atomoc, 2019-11-12:
Надо пример с GyverButton добавить в пример GTimer_manual, либо типа того. Чтобы по нажатию на кнопку запускался setPeriod(3000) а уже в if (isReady()) выводилось: Serial.println("Timer!");

AlexGyver, 2019-11-12:
Не понял. Опиши словами логику работы кода, без привязки к библиотекам. Чисто что как должно работать

atomoc, 2019-11-12:
GyverButton может запутать тех, кто ещё до него не дошёл

atomoc, 2019-11-12:
Я пытаюсь пример поправить и сделать ПР, примешь его потом, как допилишь код. Сам подумаю тогда, как его лучше оформить.

AlexGyver, 2019-11-12:
Да фиг с ним с примером, ты задачу опиши, какую нужно решить. Мне кажется что то ты не так делаешь просто

atomoc, 2019-11-12:
Жмёшь на кнопку, запускается setTimeout(3000), после заданного интервала срабатывает true в isReady и никакого цикла далее не следует. Ты же вроде как понял

atomoc, 2019-11-12:
Или нужно описать логику для общественности?

atomoc, 2019-11-12:
По моему пример как раз таки и решил эту задачу

AlexGyver, 2019-11-12:
Я понял чего не хватало для счастья, но зачем оно было - понял только сейчас. Я бы просто ввел ещё один бул флаг в основной программе) но да, с доработанной либой будет красивее

atomoc, 2019-11-12:
https://www.arduino.cc/en/tutorial/button не хочется брать отсюда код для описания примера, по этому и завёл речь о GyverButton

AlexGyver, 2019-11-12:
Да можно примеров с кнопками и таймерами много придумать)

atomoc, 2019-11-12:
Если будет setTimeout буду ещё более очень очень признателен, да и вряд ли я один

AlexGyver, 2019-11-12:
Сделаю, это ерунда. По доработке этой либы вообще ничего не просили ниразу

atomoc, 2019-11-12:
Низкий поклон и наилучших пожеланий

AlexGyver, 2019-11-12:
А isReady не трогать, само имя? А то не знаю как у вас там в js

AlexGyver, 2019-11-12:
в общем выкатил 3.0, вообще всё переделал и оптимизировал, теперь это полноценный таймер со всеми плюшками =)

atomoc, 2019-11-12:
Моему ликованию нет предела, спасибо просто огромнейшая, сейчас же перепишу везде у себя код!)

Добавь ещё, пожалуйста, isEnabled(), чтобы избежать создания переменных для отслеживания состояний таймеров.

atomoc, 2019-11-12:
Или isRun()

AlexGyver, 2019-11-12:
сделал

atomoc, 2019-11-12:
Ты вообще спишь?!)

AlexGyver, 2019-11-12:
Сегодня сбил режим сильно

atomoc, 2019-11-12:
isReady() прерывает loop

AlexGyver, 2019-11-12:
> isReady() прерывает loop

Библиотека не использует прерывания

atomoc, 2019-11-12:
и вначале, если даже не задан таймаут, isReady срабатывает. То есть таймер, видимо, в режиме интервала запускается. Думаю, надо по setTimeout и setInterval лишь делать старт

atomoc, 2019-11-12:
Хотя для поддержки кода со старым таймером выгодно оставить это для тех мест, где интервал больше 1, полагаю

atomoc, 2019-11-12:
> Библиотека не использует прерывания

Постараюсь проверить вдоль и поперёк

AlexGyver, 2019-11-12:
У меня абсолютно везде таймер как интервал используется с самого запуска. Но раз полностью изменилась реализация таймера, можно сделать его остановленным при создании объекта 
Про прерывает луп не понял

AlexGyver, 2019-11-12:
> > Библиотека не использует прерывания
> 
> Постараюсь проверить вдоль и поперёк

Просто расшифруй что именно означает "прерывает loop"

atomoc, 2019-11-12:
> можно сделать его остановленным при создании объекта

да, только оставить для тех мест, где задано время при создании экземпляра GTimer_ms timer(1000)

[27 more replies]

closed 2019-11-12 by atomoc