AlexGyver/GyverLibs #17: GyverButtons

x3mEr, 2019-11-16:
Version: 2.14 (09-sep-2019)
Holding a button resets variable "last_counter", thereby reseting isSingle, isDouble, isTriple and other (last_counter).
Баг или фича: принцип работы библиотеки такой, что различные нажатия кнопок сбрасываются только после вызова опрашивающей функции (isClick, isHolded...).
Но isSingle, isDouble and isTriple и остальные (last_counter) можно сбросить, если удерживать кнопку (hold).
Баг бага или баг фичи: такой сброс последовательности нажатий удержанием кнопки не сбрасывает флаг counter_flag.

AlexGyver, 2019-11-19:
Это интересно. Вообще это можно использовать как фичу для алгоритма клик-клик-удержание

x3mEr, 2019-11-20:
Если исправить библиотеку, чтобы счётчик нажатий last_counter (вроде так, не помню уже) не сбрасывался в атрибуте isHolded, то её тоже можно будет использовать для такого алгоритма ;)
Но надо смотреть, что такая правка может задеть.

x3mEr, 2019-11-20:
*нажал мимо кнопки)

AlexGyver, 2019-11-20:
Надо попробовать)

AlexGyver, 2019-12-14:
выкатил 3.1 версию, посмотри пример clicks_step

x3mEr, 2019-12-14:
Так и не скажешь. Вредо всё круто.
То, о чём писал я - исправлено)

В GyverButton.cpp в стороке 121 зачем нужно условие (millis() - btn_timer >= _step_timeout)?

AlexGyver, 2019-12-15:
Затем что так работает "шаг", у него свой таймер

x3mEr, 2019-12-15:
done!

closed 2019-12-15 by x3mEr

x3mEr, 2019-12-15:
> Затем что так работает "шаг", у него свой таймер
Понял. Свой определитель isHold.