AlexGyver/GyverLibs #18: Double click is always set single click flag.

x3mEr, 2019-11-16:
Double click is always set single click flag.
Double, triple (and more) click always includes single click. For example, after double clicking both functions isDouble() and isClick() will return true.

Двойное и более нажатие всегда подразумевает и одинарное. Т. е. при, например, двойном нажатии кнопки, обе функции isClick() и isDouble() вернут true.

AlexGyver, 2019-11-19:
Да, именно для этого есть isSingle, чтобы ловить один клик отдельно от двойного. В библиотеке не просто так много разных методов

x3mEr, 2019-11-20:
Т. е. совет при использовании библиотеки: в программах, где планируется различать двойные и одинарные нажатия, лучше использовать только isSingle и isDouble, без isClick?

AlexGyver, 2019-11-20:
isClick легче и быстрее, если в программе нужна только отработка клика и удержания, то юзать isClick. А если множественные нажатия, то isSingle и его друзей. Также ещё есть isPress и release, которые тоже можно юзать. Но вообще есть над чем подумать, отделить флаг клика. Ещё есть вариант сделать тик булевым, и уже по нему отслеживать были ли изменения

closed 2020-05-22 by AlexGyver