AlexGyver/GyverLibs #19: GButton / Не срабатывает isSingle в режиме AUTO

atomoc, 2019-11-17:
С isDouble и остальными по моему так же. В общем без MANUAL и tick() в loop никак.

x3mEr, 2019-11-17:
После setTickMode(true) не работает?

Пожелание Алексу -
1) в библиотеках писать дефолтные значения в пояснениях к функциям;
2) писать какие значения аргументов за что отвечают. Например, в void setTickMode(uint8_t tickMode) - false для MANUAL и true AUTO.

> 
> 
> С isDouble и остальными по моему так же. В общем без MANUAL и tick() в loop никак.

После setTickMode(true) не работает?

Пожелание Алексу -
1) в библиотеках писать дефолтные значения в пояснениях к функциям;
2) писать какие значения аргументов за что отвечают. Например, в void setTickMode(uint8_t tickMode) - false для MANUAL и true AUTO.

atomoc, 2019-11-17:
> После setTickMode(true) не работает?

Работает, но так же как и с AUTO, то есть раз работает, раз нет, два раза работает, 5 нет...

isClick() всегда отрабатывает.

x3mEr, 2019-11-17:
> 
>     2. писать какие значения аргументов за что отвечают. Например, в void setTickMode(uint8_t tickMode) - false для MANUAL и true AUTO.

Разобрался, сам не допонял.

По поводу проблемы... Думаю, ошибка в этом (разберу не примере isSingle()):
Для получения true, необходимо выполнения двух условий counter_flag == true и last_counter == 1.
В случае tickMode == AUTO, isSingle() вызывает обработчик нажатий GButton::tick().
Он, в свою очередь, изменяет counter_flag и last_counter только по условию (millis() - btn_timer >= _click_timeout. Вот этот _click_timeout больше времени выполнения предыдущего кода, т. е. условие не выполняется и нужные для isSingle() значения не меняются в момент его вызова.
В случае tickMode == MANUAL вручную выполняется где-то в коде tick(), потом где-то в коде выполняется опрос isSingle()... Между их выполнениями проходит больше времени, чем _click_timeout (НО! это не всегда, надо проверять (например, два подряд идущих вызова) и контролировать... Блин, снова у себя всё править:( ), соответственно, к моменту вызова isSingle() все переменные успеют обновиться.

AlexGyver, 2019-11-19:
Да, авто режим не очень корректно работает, потому что очень много завязано на счёте времени. Но в локальных циклах все будет работать

closed 2020-05-22 by AlexGyver