AlexGyver/GyverLibs #20: Ошибка экспорта в create.arduino.cc

mshavliuk, 2019-11-18:
## Проблема

При попытке загрузки библиотек в облачный редактор Arduino-скетчей вылетает следующая ошибка:

```
Libraries that could not be uploaded:

[] library.properties not found
```

Причина в отсутствии файла `library.properties`, подробности описаны [в соответствующем форуму](https://forum.arduino.cc/index.php?topic=613965.0).

## Возможное решение

Добавить в каждую библиотеку файл `library.properties` с соответствующей схемой данных: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Шаблон файла:
```
name=Gyver____
version=___
author=AlexGyver <beragumbo@ya.ru>
maintainer=AlexGyver <beragumbo@ya.ru>
sentence=____
paragraph=___
category=___
url=https://github.com/AlexGyver/GyverLibs/tree/master/___
architectures=*
```

mshavliuk, 2019-11-18:
Создал набросок пулл-реквеста - планирую продолжить на выходных.

AlexGyver, 2019-11-19:
Добавлю во все либы, давно пора :)

closed 2019-12-14 by AlexGyver