AlexGyver/GyverLibs #23: GyverTimer012

fordisimo, 2019-12-11:
Dear Alex.
In the example to the library, only Timer1 is used.
Have you checked Timer2 and Timer0.
For some reason, only Timer0 works for me.

Request placed on your forum
https://community.alexgyver.ru/threads/problema-v-ispolzovanii-biblioteki-gyvertimer012.2166/

Look here please.
Teska.

Уважаемый, Алекс.
В примере к библеотеке использован только Timer1.
Проверяли ли Вы Timer2 and Timer0.
У меня почему то работает только Timer0.

Запрос поместил на Вашем форуме
https://community.alexgyver.ru/threads/problema-v-ispolzovanii-biblioteki-gyvertimer012.2166/

Посмотрите пожалуйста.
Теска.

AlexGyver, 2019-12-14:
Привет, действительно, релизная версия перестала работать. Починим

fordisimo, 2019-12-14:
Спасибо, причина именно в этом - использовал не недопустимое значение переменной.

fordisimo, 2019-12-14:
Еще раз огромное спасибо за оперативную реакцию.
Александр.

closed 2019-12-14 by fordisimo