AlexGyver/GyverLibs #24: TM1637. Не управляются точки на дисплее.

fordisimo, 2019-12-25:
Суть проблемы : - Заменил дисплей с ":" по середине на дисплей с Десятичными точками.
- Обнаружил, что точки не управляются.
Просьба к авторам библиотеки рассмотреть проблему и добавить/расширить функции библиотеки.

Ниже привожу быстрое решение, которое из-за лимита времени применил. Оно справедливо только для функции point() и display().
Подчеркиваю, что с огромным уважение отношусь к авторам библиотеке и позволил себе внести изменения только из-за крайнего цейтнота.
(в файлах библиотеки отмечены строки, в которых произведены изменения).

Пример (все как для Оригинальной библиотеки + небольшие изменения):

#include <GorTM1637.h>

#define CLK A5 //pin-ы для TM1637
#define DIO A4

GorTM1637 disp(CLK, DIO);

void setup()
{
disp.clear(); // инициализация TM1637
disp.brightness(7); // ярк, 0 - 7 (минимум - максимум)
}

void loop()
{
disp.point(POINT_ON,2); // ТОЧКУ поставим во 2-ю поз. ( слева направо, начиная с нуля)

disp.displayInt(301); // получим па дисплее 30.1

delay(2000);

disp.point(POINT_OFF,2); // погасить точку (если надо)
disp.displayInt(103); // получим па дисплее 103

delay(2000);

}

Библиотека с модификациями в посте здесь https://community.alexgyver.ru/threads/biblioteka-dlja-tm1637-s-desjatichnoj-tochkoj.2249/

AlexGyver, 2020-02-21:
а для какого дисплея возник вопрос? Можно ссылку?

fordisimo, 2020-02-25:
Дисплей куплен в магазине,  контроллер ТМ 1637 (выдно на фото).
С уважением, Александр.
[image: TM1637_1.jpg]
[image: TM1637_2.jpg]
[image: TM1637_3.jpg]

AlexGyver, 2020-02-25:
не вижу фоток

fordisimo, 2020-02-26:
а приатаченые файлы есть?

AlexGyver, 2020-02-26:
нету

fordisimo, 2020-02-27:
так должно работать :)
https://drive.google.com/drive/folders/1B9COL8aOvwkxxeJFliNkD4Yy_FW9sLhr?usp=sharing

AlexGyver, 2020-02-27:
Поддержки такого дисплея в библиотеке нет. Можно ссылочку? Куплю себе

fordisimo, 2020-02-28:
https://aliexpress.ru/item/32962489291.html?spm=a2g0s.9042311.0.0.1ce233edAsKNcl

AlexGyver, 2020-02-28:
Окей босс

fordisimo, 2020-03-02:
🤔

fordisimo, 2020-03-03:
https://community.alexgyver.ru/threads/biblioteka-dlja-tm1637-s-desjatichnoj-tochkoj.2249/

тут я быстренько   слепил заплатку, глубоко не прорабатовал.

пн, 2 бер. 2020 о 10:16 Olexander Horoshko <fordisimo@gmail.com> пише:

AlexGyver, 2020-03-03:
Окей, как приедет сделаю что нибудь приличное

fordisimo, 2020-03-05:
удачи

Anymorf, 2021-01-31:
как там есть сподвижки по точкам?

AlexGyver, 2021-01-31:
Не, нет времени

AlexGyver, 2021-03-15:
Идёт разработка

closed 2021-03-15 by AlexGyver