AlexGyver/GyverLibs #28: GyverEncoder / Заменить таймер дребезга на микросекунды

Uinden, 2020-01-25:
У меня энкодеры 2го типа (6.5мм).
Припаяны проводами к ардуине на пины с прерываниями. tick() повешан на прерывания. (Без прерываний та же проблема) (Пробовал 2 энкодера).
При вращении код через раз неправильно определяет направление вращения.
После замены https://github.com/AlexGyver/GyverLibs/blob/master/GyverEncoder/GyverEncoder.cpp#L181
```
uint32_t thisMls = millis();
```
На
```
uint32_t thisMls = micros();
```
И https://github.com/AlexGyver/GyverLibs/blob/master/GyverEncoder/GyverEncoder.h#L63
```
#define ENC_DEBOUNCE_TURN 1
```
На
```
#define ENC_DEBOUNCE_TURN 10
```
Всё заработало. Нужно очень быстро крутануть по часовой стрелке, чтоб направление определилось неправильно. Против часовой стрелки вообще работает идеально.
Остались проблемы с количеством - вместо 10 срабатываний иногда регистрирует 8-11. 
Возможно у меня совсем палёные энкодеры. Но было бы неплохо переделать на микросекунды для таких энкодеров.

AlexGyver, 2020-02-21:
Странно, а какой энкодер? Голый, без цепей антидребезга?

Uinden, 2020-02-21:
@AlexGyver 
Энкодер 6.5мм с 5ю выводами. Без надписей. Из примечаний - боковые контакты широкие и будто обрезаны на заводе.
Провода припаивал напрямую от энкдодера к пинам китайской Arduino Nano (на atmega 328p в QFN-32 корпусе)
[image]
[image]
[image]
[image]
[image]

AlexGyver, 2020-02-21:
А, голый энк. Вообще для вращения человеком достаточно антидребезга в пару тройку мс. Версия библиотеки какая? Какой алгоритм выбран?

Uinden, 2020-02-22:
Сегодня скачал последнюю версию из репозитория, использую код без модификаций https://github.com/AlexGyver/GyverLibs/blob/master/GyverEncoder/examples/encoder_demo/encoder_demo.ino
Попытался записать состояния пинов в виде кода, чтоб залить на другую ардуину и проиграть первой, но не получается.
Но код есть. За одно вращение энкодера выглядит он так:
```C++
digitalWrite(3, LOW);
delayAuto(2308);
digitalWrite(2, LOW);
delayAuto(5688);
digitalWrite(3, HIGH);
delayAuto(404);
digitalWrite(2, HIGH);
delayAuto(5904);
digitalWrite(3, LOW);
delayAuto(2224);
digitalWrite(2, LOW);
delayAuto(2952);
digitalWrite(3, HIGH);
delayAuto(344);
digitalWrite(2, HIGH);
delayAuto(2404);
digitalWrite(3, LOW);
delayAuto(408);
digitalWrite(2, LOW);
delayAuto(8);
digitalWrite(3, HIGH);
delayAuto(12);
digitalWrite(2, HIGH);
delayAuto(2368);
digitalWrite(3, LOW);
delayAuto(8);
digitalWrite(2, LOW);
delayAuto(8);
digitalWrite(3, HIGH);
delayAuto(12);
digitalWrite(3, LOW);
delayAuto(8);
digitalWrite(2, HIGH);
delayAuto(12);
digitalWrite(3, HIGH);
delayAuto(16);
digitalWrite(3, LOW);
delayAuto(16);
digitalWrite(3, HIGH);
delayAuto(8);
digitalWrite(2, LOW);
delayAuto(12);
digitalWrite(2, HIGH);
delayAuto(8);
digitalWrite(3, LOW);
delayAuto(16);
digitalWrite(3, HIGH);
delayAuto(16);
digitalWrite(3, LOW);
delayAuto(16);
digitalWrite(3, HIGH);
delayAuto(16);
digitalWrite(3, LOW);
delayAuto(8);
digitalWrite(2, LOW);
delayAuto(8);
digitalWrite(3, HIGH);
delayAuto(12);
digitalWrite(3, LOW);
delayAuto(8);
digitalWrite(2, HIGH);
delayAuto(12);
digitalWrite(2, LOW);
delayAuto(8);
digitalWrite(3, HIGH);
delayAuto(8);
digitalWrite(2, HIGH);
delayAuto(12);
digitalWrite(2, LOW);
delayAuto(16);
digitalWrite(2, HIGH);
delayAuto(8);
digitalWrite(3, LOW);
delayAuto(16);
digitalWrite(3, HIGH);
delayAuto(1248);
digitalWrite(3, LOW);
delayAuto(8);
digitalWrite(2, LOW);
delayAuto(8);
digitalWrite(3, HIGH);
delayAuto(8);
```
Дребезг сумасшедший.
Вот код которым я это генерировал https://pastebin.com/raw/3a2UvRbC
И вот я пытался его залить на вторую ардуину https://pastebin.com/raw/bwuS4VsZ
Видимо при проигрывании эти короткие импульсы проскакивают мимо. Но проблема наверное в них.

AlexGyver, 2020-02-23:
Судя по коду, энкодер надо в помойку. Он не должен так работать

Uinden, 2020-02-23:
Я их 5 штук купил. Все 5 попробовал, все 5 так криво работают.
Достал китайский логический анализатор и там действительно всё плохо. Очень много шумов.
[image]
Видимо issue можно закрывать. Буду думать как фильтровать этот дребезг.

Uinden, 2020-02-23:
Вообще мне кажется я понял в чём проблема библиотеки с моими энкодерами.
Сигнал притягивается к земле и проходит какое-то время. Потом вдруг возникает скачек до VCC на пару микросекунд и условие из библиотеки ```(debounceDelta > ENC_DEBOUNCE_TURN)``` валидное. И она регистрирует изменение состояния пина.
Хотя надо бы подождать и посмотреть как долго будет сигнал HIGH на ноге. Если это меньше Х микросекунд, то игнорировать его.
По крайней мере мне так кажется.
Но я уже сам попробую это сделать.

AlexGyver, 2020-02-23:
Да, у меня антидребезг сделан чуть иначе, несовсем корректно. На странице с описанием этой библиотеки у меня на сайте есть схемы антидребезга на rc цепях, все очень тривиально и просто. Конечно такие вещи лучше аппаратно фильтровать, над переделкой алгоритма антидребезга я подумаю мб переделаю

Uinden, 2020-02-23:
RC схема не помогла. Она, видимо, полезна для кнопок, когда после изначального контакта есть небольшой дребезг. А тут дребезг местами появляется посреди подтяжки к LOW. 
На анализаторе сигналы стали чище, но единичные импульсы проскакивают и библиотека их регистрирует.
Я в C++ не очень, но накидал такую лапшу
```C++
#define CLK 2
#define DT 3

#define DEBOUNCE_US 100 // сколько микросекунд ждать перед передачей состояния пина в tick()

#include "GyverEncoder.h"
Encoder enc1(0, 0, ENC_NO_BUTTON, TYPE2);  // для работы без кнопки и сразу выбираем тип

unsigned long CLKlastChange; // последнее изменение состояния пина
bool CLKlastState; // последнее состояние пина (не стабилизировано, может быть дребезгом)
bool CLKreportedState; // последнее состояние пина которое мы сообщили в библиотеку и которое стабилизировано (CLKlastState может быть дребезгом)

unsigned long DTlastChange;
bool DTlastState;
bool DTreportedState;

void setup() {
  Serial.begin(115200);
  CLKlastState = digitalRead(CLK);
  DTlastState = digitalRead(DT);

  CLKreportedState = CLKlastState;
  DTreportedState = DTlastState;

  CLKlastChange = micros();
  DTlastChange = micros();

  attachInterrupt(0, isrCLK, CHANGE);
  attachInterrupt(1, isrDT, CHANGE);

  enc1.tick(CLKlastState, DTlastState, LOW); // тестовый скрипт, кнопка игнорируется и всегда LOW
}

void loop() {
  unsigned long currentMicros = micros();
  unsigned long delta;
  // если состояние пина стабилизировано и мы его ещё не сообщали, сообщаем
  delta = currentMicros-CLKlastChange;
  // видимо прерывание на лету меняет CLKlastChange и переменная улетает выше 4000000000
  if(delta > DEBOUNCE_US && delta < 4000000000 && CLKlastState != CLKreportedState) {
    // В CLK передаём текущее состояние, в DT последнее переданное (DTlastState может быть дребезгом)
    enc1.tick(CLKlastState, DTreportedState, LOW); 
    CLKreportedState = CLKlastState;

    Serial.print(currentMicros);
    Serial.print(") reported CLK ");
    Serial.print(CLKreportedState);
    Serial.print(" stabilized in ");
    Serial.println(delta);
  }
  delta = currentMicros-DTlastChange;
  if(delta > DEBOUNCE_US && delta < 4000000000 && DTlastState != DTreportedState) {
    enc1.tick(CLKreportedState, DTlastState, LOW); 
    DTreportedState = DTlastState;

    Serial.print(currentMicros);
    Serial.print(") reported DT ");
    Serial.print(DTreportedState);
    Serial.print(" stabilized in ");
    Serial.println(delta);
  }

  if (enc1.isRight()) Serial.println("Right");         // если был поворот
  if (enc1.isLeft()) Serial.println("Left");
}

void isrCLK() {
  // сбрасываем таймер и запоминаем новое НЕ СТАБИЛИЗИРОВАННОЕ состояние
  CLKlastState = digitalRead(CLK);
  CLKlastChange = micros();
}
void isrDT() {
  DTlastState = digitalRead(DT);
  DTlastChange = micros();
}
```
В библиотеке отключил определение дребезга ```#define ENC_DEBOUNCE_TURN 0``` и всё заработало оооочень стабильно.
Всё ещё проскакивают какие-то неадекватные срабатывания
```
40191484) reported DT 0 stabilized in 108
40194840) reported DT 1 stabilized in 680
40198832) reported DT 0 stabilized in 104
40202172) reported DT 1 stabilized in 252
40205804) reported DT 0 stabilized in 512
40209460) reported DT 1 stabilized in 584
40213112) reported DT 0 stabilized in 436
40217044) reported DT 1 stabilized in 128
40222076) reported DT 0 stabilized in 108
40224080) reported DT 1 stabilized in 668
40227732) reported DT 0 stabilized in 1804
40231472) reported DT 1 stabilized in 2828
```
**Хотя я в анализаторе такого не вижу.** Не вижу чтоб какой-то пин подскочил с LOW до HIGH на 100+ микросекунд. Может это мой косяк в коде.
Но в любом случае библиотека на них не реагирует. Всё стало очень хорошо.
Пропусков нет, неверных определений вращений очень мало. Может 1 из 200.

Но тут есть не очевидная проблема - без прерываний оно не работает. Если в начале ```loop()``` опрашивать пины, то пока код работает время уже пройдёт и на следующем вызове состояние определится как стабилизированное. 
В общем я буду этот код для себя использовать, но решение так себе.

Разобрал один энкодер, там китайцы всё смазкой залили. Наверняка это она виновата.
[image]
[image]

AlexGyver, 2020-02-24:
Может почистить тогда энк?) Потому что "круглый" модуль с rc цепями как на схеме на сайте работает просто изумительно, даже без программного гашения работает

Uinden, 2020-02-24:
Я, конечно, могу. Но наверняка не я один такие энкодеры получил.
Я ещё с анализатором поигрался и добавил вывод.
При вызове `enc1.tick` за одно делаю `digitalWrite` на соседние пины, чтоб увидеть что получается.
`DEBUG_CLK` и `DEBUG_DT` при `#define DEBOUNCE_US 70`
[image]
Не всегда срабатывает. Слишком много шумов.
Получается нужно делать код ещё хитрее, чтоб он дребезг принимал за LOW. Но это для меня уже слишком, да и так всё хорошо работает. (Раньше у меня половина вращений неправильно определялась, а сейчас просто сказка.)

Хотя это может быть вполне хорошее решение, которое будет работать даже через loop. Если состояние пина меняется чаще 1 раза за 300мс (столько у меня получилось при очень быстром рывке рукой) то это LOW.
Я на всякий случай поменяю заголовок issue чтоб люди через поиск могли найти на случай если у них такая же проблема будет, может нас много таких.

AlexGyver, 2020-02-24:
Делать digitalWrite совсем нельзя, на стандартном ядре эта функция отнимает очень много процессорного времени. Если шумы будут дергать digitalWrite - вообще ничего не будет работать. Нужно собрать цепи антидребезга как на схеме, всё будет работать. Возможно есть смысл почистить энкодер от лишней смазки или что это там такое

Uinden, 2020-02-24:
digitalWrite там для отладки. Он дёргается только когда в tick() передаются стабилизированные значения без дребезга что бы видеть что передаётся в библиотеку.
RC схему с сайта я собирал (2 резистора по 10кОм и 2 кондёра по 100nF) и это не помогло. Стало чище, но единичные пики проскакивают посреди LOW прохода и библиотека их принимает.
Я закрою issue и по идее если ещё люди будут с такой проблемой, они смогут отписаться и уже можно будет думать над изменением кода в библиотеке.

closed 2020-02-24 by Uinden