AlexGyver/GyverLibs #29: Не хватает функции в GyverButton

planer-pro, 2020-01-29:
А конкретно функции, которая бы чистила все буфера butt.tick(); о захвате нажатия, а то в не зависимости нужно слушать в данный момент остальные кнопки помимо тех которые нужно, tick все запоминает в буфер а потом делает ложное срабатывание, когда остальным кнопкам уже разрешено реагировать, но ты еще ничего не нажимал. Пример с постраничным меню дисплея, на каждой странице разрешено читать свои кнопки, а не все сразу и писать в буфер что они тоже были нажаты - прочитал butt.isClick(); и сразу все буфера clear а еще лучше выбранные буфера.

AlexGyver, 2020-02-21:
Данная фича была добавлена в версии 3.4, метод resetStates()

planer-pro, 2020-02-21:
good!

closed 2020-02-21 by AlexGyver