AlexGyver/GyverLibs #34: Compiling error MicroLED Matrix

derdek, 2020-02-26:
In file included from C:\Users\Derdek\Documents\Arduino\libraries\microLED/microLED.h:40:0,
                 from lab_2.ino:1:
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:43:18: error: variable or field 'WS2812B_sendData' declared void
 #define PTR_TYPE uint8_t*
                  ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:47:24: note: in expansion of macro 'PTR_TYPE'
 void WS2812B_sendData (PTR_TYPE data, int16_t datlen, uint8_t maskhi, uint8_t *port, uint8_t *portreg, uint8_t bright) {
                        ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:43:18: error: 'uint8_t' was not declared in this scope
 #define PTR_TYPE uint8_t*
                  ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:47:24: note: in expansion of macro 'PTR_TYPE'
 void WS2812B_sendData (PTR_TYPE data, int16_t datlen, uint8_t maskhi, uint8_t *port, uint8_t *portreg, uint8_t bright) {
                        ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:47:33: error: 'data' was not declared in this scope
 void WS2812B_sendData (PTR_TYPE data, int16_t datlen, uint8_t maskhi, uint8_t *port, uint8_t *portreg, uint8_t bright) {
                                 ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:47:39: error: 'int16_t' was not declared in this scope
 void WS2812B_sendData (PTR_TYPE data, int16_t datlen, uint8_t maskhi, uint8_t *port, uint8_t *portreg, uint8_t bright) {
                                       ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:47:55: error: 'uint8_t' was not declared in this scope
 void WS2812B_sendData (PTR_TYPE data, int16_t datlen, uint8_t maskhi, uint8_t *port, uint8_t *portreg, uint8_t bright) {
                                                       ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:47:71: error: 'uint8_t' was not declared in this scope
 void WS2812B_sendData (PTR_TYPE data, int16_t datlen, uint8_t maskhi, uint8_t *port, uint8_t *portreg, uint8_t bright) {
                                                                       ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:47:80: error: 'port' was not declared in this scope
 void WS2812B_sendData (PTR_TYPE data, int16_t datlen, uint8_t maskhi, uint8_t *port, uint8_t *portreg, uint8_t bright) {
                                                                                ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:47:86: error: 'uint8_t' was not declared in this scope
 void WS2812B_sendData (PTR_TYPE data, int16_t datlen, uint8_t maskhi, uint8_t *port, uint8_t *portreg, uint8_t bright) {
                                                                                      ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:47:95: error: 'portreg' was not declared in this scope
 void WS2812B_sendData (PTR_TYPE data, int16_t datlen, uint8_t maskhi, uint8_t *port, uint8_t *portreg, uint8_t bright) {
                                                                                               ^
C:\Users\Derdek\Documents\Arduino\libraries\microLED/ws2812_send.h:47:104: error: 'uint8_t' was not declared in this scope
 void WS2812B_sendData (PTR_TYPE data, int16_t datlen, uint8_t maskhi, uint8_t *port, uint8_t *portreg, uint8_t bright) {
                                                                                                        ^
Ошибка компиляции.

q7c, 2020-03-20:
для какой платы?

derdek, 2020-03-22:
Arduino leonardo

AlexGyver, 2020-12-21:
библиотека обновлена

closed 2020-12-21 by AlexGyver