AlexGyver/GyverLibs #35: Лента жарит яки чёрт

Crazy-Geeks, 2020-02-29:
Atmega 32u4 (Pro Micro), полёт нормальный, но есть, как говорится, один нюанс. setMaxCurrent не работает. Ток стоит 1А, лента выжигает глаза буквально, в FastLED при ограничении в 1А она еле светила
[image]
[image]
(Через камеру не передашь, естественно)

AlexGyver, 2020-03-11:
А что за лента? По моим тестам ограничение тока работало прям плюс минус пара миллиампер. Но ленты разные...

Crazy-Geeks, 2020-03-11:
WS2812B, у giant4 брал
Камень 32u4 16 mhz
Такое ощущение, что эта функция вообще игнорируется, при изменении значений яркость так и остаётся 
У FastLED то оно работает, поэтому проблемы не с лентой

AlexGyver, 2020-03-11:
А какая версия либы? И ты ток измерял, или чисто визуально?

AlexGyver, 2021-03-15:
Библиотека обновлена

closed 2021-03-15 by AlexGyver