AlexGyver/GyverLibs #44: uart.write does not work, but works when change to uart.println

eugenebartosh, 2020-05-21:
Hi Alex, 

I have setup your core and used uart lib instead of Serial. 
Below is the code, it is quite straight-forward. I get analog input through interupt and send it to UART (uart.write). It does sends nothing (in monitor). But when I change it to uart.println - it works. 
Any ideas? 

Thanks, Eugene 

#define switch1 4               
#define switch2 3               

boolean sw1, sw2, lsw1, lsw2;
uint8_t v0, v1; 
uint16_t lastADC, val; 
boolean adcf = false; 
uint8_t dv1 = 1, dv2 = 1; 
uint8_t pos = 0; 
    
void setup() {
  pinMode(switch1, INPUT);
  digitalWrite(switch1, HIGH);  // enable internal pullup
  pinMode(switch2, INPUT);
  digitalWrite(switch2, HIGH);
  ADMUX =  B00000000;           // set external aref and port
//  ADMUX =  B01000000;         // выбор внутреннего опорного напряжения (Vcc - 5V) и порта 5 (A0)
  ADCSRA = B10101100;           // ADC enable, interrupt enable, set default prescaler
  ADCSRB = B00000000;           // free running mode
  sei();            // enable interrupts
  ADCSRA |=B01000000;           // start first conversion
  uart.begin(1000000);
  lsw1 = digitalRead(switch1);
  lsw2 = digitalRead(switch2);
  sw1 = lsw1; 
  sw2 = lsw2; 
  if ( ( sw1 ) && ( !sw2 ) ) { dv1 = 1; dv2 = 2; } 
  else if ( ( !sw1 ) && ( sw2 ) ) { dv1 = 5; dv2 = 6; } 
  else if ( ( !sw1 ) && ( !sw2 ) ) { dv1 = 1; dv2 = 4; } 
  else { dv1 = 1; dv2 = 1; } 
}

void loop() { 
  if ( adcf ) { 
    cli();
    val = lastADC; 
    adcf = false;
    sei();
    switch ( dv2 ) {  
      case 1 : v0 = val; break; 
      case 4 : v0 = val >> 2; break;
      case 2 : v0 = val >> 1; break; 
      case 6 : v0 = val * dv1 / dv2; 
    } 
    pos = pos + 1;
    switch ( pos ) { 
      case 1 : v1 = v0 >> 4; break; 
      case 2 : v1 = v1 | (v0 & B11110000); break;
      case 3 : uart.write(v1); pos = 0; break;
    } 
  } 
}

ISR(ADC_vect) {
  lastADC = ADC; 
  adcf = true;
}

AlexGyver, 2020-05-22:
Hello! uart.write(byte) is working correctly!
`#include <GyverUART.h>

void setup() {
  uart.begin(9600);
  uart.println("Hello!");
  uart.write(35);
  
}

void loop() {

}`
results

> Hello!
> #

AlexGyver, 2020-05-22:
Hello! I've fixed it, thanks!

closed 2020-05-22 by AlexGyver

eugenebartosh, 2020-05-23:
спасибо, Alex! А не подскажете как мне обновить до выхода релиза? я не гуру, покопался но с ходу не нашёл где можно подложить свежий файл

AlexGyver, 2020-05-23:
Gyveruart вышло вчера обновление, по той же ссылке скачивается

eugenebartosh, 2020-05-23:
сорям я наверно что-то не понимаю. Здесь на GitHub я вижу только обновлённый исходник, последние релизы Core и Libs - в марте 2020

AlexGyver, 2020-05-23:
Gyveruart обновился в этом репозитории, обновился также релиз. О GyverCore вроде речь не шла

eugenebartosh, 2020-05-24:
ах да сорри, не заметил что релиз в папке с исходником лежит, спасибо!

AlexGyver, 2020-05-24:
Всм в папке? Сам репозиторий - актуальные исходники, релизы лежат в гитхабовском списке релизов, прямые ссылки на загрузку есть в описании репозитория где список и описание всех либ

eugenebartosh, 2020-05-24:
самостоятельно невозможно разобраться в структуре - я просматривал список релизов здесь https://github.com/AlexGyver/GyverLibs/releases, по ссылке releases из кода, там только лежит старый релиз GyverUART от 2019 г.

AlexGyver, 2020-05-25:
Дата там не меняется, меняется код версии. Актуальная 1.8 сейчас