AlexGyver/GyverLibs #47: Может ли библиотека GyverButton отслеживать такой тип сигнала?

Railnolds, 2020-06-09:
Добрый день.
У меня вопрос по библиотеке GyverButton. Хочу повесит вместо кнопки на пин сигнал с аппаратуры радиоуправления RadioLink RC6GS. На самой аппаратуре есть кнопка без фиксации. Хочу сделать чтобы при нажатии на кнопку аппаратуры сигнал от приемника приходил на контроллер, а библиотека могла бы распознавать сколько нажатий было сделано и соответственно включать разные функции (тоесть далее работала как обычно).
Сигнал с приемника аппаратуры я могу отслеживать и выводить в монитор порта следующим образом:
```
ppm_str = pulseIn (PIN, HIGH);
Serial.println(ppm_str);
```
Этот сигнал выдается в порт числами от 1000 до 2000. Когда кнопка не нажата - 1000, если нажата - 2000.
Вопрос: можно ли как-то сделать, чтобы библиотека принимала и распознавала такой сигнал? Может его надо как-то преобразовать, чтобы библиотека считала это нажатием простой физической кнопки.
Буду очень признателен за помощь.

faunel, 2020-06-24:
Подсоединяете вход кнопки к выходу свободного порта. По умолчанию выход порта делаете на высокий уровень
И далее проверяете

Пример.

pinMode(12, OUTPUT);
digitalWrite(12, HIGH);
delay(10);
if (ppm_str == 2000){ // или так if (ppm_str > 1000){
    digitalWrite(12, LOW);
}

12 замените на номер порта, который будет подключен к входу кнопки

Railnolds, 2020-06-26:
Я сделал по другому, теперь работает как надо

ppm_light = pulseIn (pinCh3, HIGH);
  butt1.tick(ppm_light > 1500);
  if (butt1.hasClicks()) clicks = butt1.getClicks();         // получение количества нажатий кнопки

  switch (clicks) {
    case 1:
    case 2:
    case 3:

closed 2021-03-15 by AlexGyver