AlexGyver/GyverLibs #52: Неопределённое поведение в функции `timer0_setPeriod`

ilyavenner, 2020-10-17:
Рассмотрим параметр `time` функции [timer0_setPeriod](https://github.com/AlexGyver/GyverLibs/blob/bab4a7ca4cbcdb649e920a7776e5fd09e5773411/GyverTimer012/GyverTimer0.cpp#L11). Так как каскад `if-else...` не обрабатывает все возможные значения аргумента `time`:
```cpp
if (time > 1 && time <= 16) {
    divisor = 1;
    divValue = 1;
} else if (time > 16 && time <= 128) {
    divisor = 8;
    divValue = 2;
} else if (time > 128 && time <= 1024) {
    divisor = 32;
    divValue = 3;
} else if (time > 1024 && time <= 4096) {
    divisor = 64;
    divValue = 4;
} else if (time > 4096 && time <= 16384) {
    divisor = 128;
    divValue = 5;
} else TCCR0B = 1;
```
то для любых значений `time` больше `16384` на следующей строке произойдёт чтение не инициализированной переменной `divisor`:
https://github.com/AlexGyver/GyverLibs/blob/bab4a7ca4cbcdb649e920a7776e5fd09e5773411/GyverTimer012/GyverTimer0.cpp#L38

P.S. Изменение типа параметра на `uint16_t` нарушит обратную совместимость.

AlexGyver, 2020-10-30:
Библиотека устарела и не используется/не обновляется

ilyavenner, 2020-11-01:
В таком случае её нужно либо удалить, либо пометить устаревшей явно.

AlexGyver, 2020-11-01:
Вроде пометил как мог

ilyavenner, 2020-11-01:
Ни в README, ни в коде таких пометок не нашёл.

AlexGyver, 2020-11-01:
[image]

AlexGyver, 2021-03-15:
Библиотека устарела

closed 2021-03-15 by AlexGyver