AlexGyver/GyverLibs #56: Конфликт GyverTimes с TFT Дисплеем на базе ST7735

SanYch86, 2020-11-04:
День добрый. Обнаружил конфликт при совместном использовании библиотеки прерываний по таймеру GyverTimers и TFT дисплея на базе ST7735. При активации канала прерываний прекращается отображение графических примитивов на дисплее (скетч ниже).
Используется arduino nano и tft oled дисплей 1.8 на ST7735. 

```
#include <Adafruit_GFX.h> // Библиотека для вывода графических примитивов на дисплей
#include <Adafruit_ST7735.h> // Библиотека драйвера дисплея

#include "GyverTimers.h" // Библиотека для работы с прерываниями по таймеру

#define TFT_CS        7
#define TFT_RST        6 
#define TFT_DC         8

Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_RST);

void setup() 
{
  tft.initR(INITR_BLACKTAB);
  tft.fillScreen(ST7735_BLACK);
  tft.setRotation(1);

  Timer1.setPeriod(40);     // Устанавливаем период таймера 
  Timer1.enableISR(); // Если закомментировать эту строку то квадрат рисуется на экране, иначе - экран залит цветом заднего фона, квадрат не рисуется!
  Timer1.stop();
}

void loop() {
  tft.fillRect(10,10,150,118,ST7735_YELLOW); // Рисуем прямоугольник заполненный желтым цветом

}
```

AlexGyver, 2020-11-27:
Привет, надо лезть смотреть что там индусы из адафрут наделали

AlexGyver, 2021-03-15:
При использовании прерываний таймера обязательно нужно объявить вектор, то есть обработчик прерывания. В твоём случае добавить
ISR(TIMER1_A) {
}
даже если он пустой. Добавлю в доку

closed 2021-03-15 by AlexGyver