AlexGyver/GyverLibs #60: Ошибка в порядке цветов в microLed

Siorinex, 2020-12-06:
Тестирую:
2812b, GRB порядок, 1 байт - нормально, красный - красный, и т.д.
2813 - параметры не меняю: красный - красный, а зеленый и синий перепутаны местами.

2813, 1 байт, что логично - переключаю на BRG, чтобы поменять зеленый и синий на места.
В результате вместо красного - зеленый, вместо зеленого - синий, вместо синего - красный!

Почесав в затылке - меняю на GBR раскладку - там норм выходит.

И вот сидишь и думаешь: в чем тогда косяк, если при перемене цветов вот такая фигня получается-то?!

AlexGyver, 2020-12-21:
Всё сделано и протестировано на всех лентах с опорой на даташиты чипов
```
// поддерживаемые чипы лент и их официальный порядок цветов:
// microLED<NUMLEDS, STRIP_PIN, MLED_NO_CLOCK, LED_WS2811, ORDER_GBR> strip;
// microLED<NUMLEDS, STRIP_PIN, MLED_NO_CLOCK, LED_WS2812, ORDER_GRB> strip;
// microLED<NUMLEDS, STRIP_PIN, MLED_NO_CLOCK, LED_WS2813, ORDER_GRB> strip;
// microLED<NUMLEDS, STRIP_PIN, MLED_NO_CLOCK, LED_WS2815, ORDER_GRB> strip;
// microLED<NUMLEDS, STRIP_PIN, MLED_NO_CLOCK, LED_WS2818, ORDER_RGB> strip;
// microLED<NUMLEDS, STRIP_PIN, MLED_NO_CLOCK, LED_WS6812, ORDER_RGB> strip;
// microLED<NUMLEDS, STRIP_PIN, CLOCK_PIN, LED_APA102, ORDER_BGR> strip;
```

AlexGyver, 2021-03-15:
Библиотека обновлена

closed 2021-03-15 by AlexGyver