AlexGyver/GyverLibs #66: Структура репозитория

anklimov, 2021-01-09:
Много полезных библиотек, проделана большая работа, но зачем они свалены в кучу в одном репозитории? Это не просто противоречит принципу git но и делает крайне неудобным использование и контрибуцию со стороны тех, кто может вносить изменения и создавать PRы
Например, мне в проекте понадобилась библиотека PID - зачем мне тянуть в проект еще полсотни? 

Ну и не надо пытаться реализовать всю логику внутри файла с расширением .h как это сделано в ряде библиотек. Можно и нужно имплементацию описывать в файлах .cpp

AlexGyver, 2021-02-02:
Помимо библиотек на аккаунте есть проекты, если репозиториев будет слишком много - я сам в этом всём запутаюсь и работать будет невозможно. Скачать одну библиотеку можно архивом из релизов. Гитом пользоваться к сожалению не умею и использую его просто как "облако", пулреквесты принимаю на почту.

"Можно и нужно имплементацию описывать в файлах .cpp" - можно и нужно, но в этом случае не работают дефайны из скетча. А на них строится большая часть оптимизации кусков кода, которую делать шаблонами будет очень некрасиво. Многие серьёзные библиотеки (не мои) реализованы в .h только ради этого

wl2776, 2021-03-14:
Поддерживаю предыдущего докладчика в части реорганизации структуры репозитория. Вопросы, поднятые @anklimov, вполне здравые.
Алекс, такой организацией репозитория Вы существенно затрудняете участие сообщества в разработке Ваших проектов.
Решать, конечно, не мне, но Вы можете получить существенную выгоду.

Да, увы, в Гитхабе сейчас возможна только плоская структура, нельзя группировать репозитории. Возможно с верией 3.0 ситуация изменится, они что-то делают. В gitlab.com это было можно с первых версий, сейчас тоже можно. Можно мигрировать туда, возможности Гитлаба гораздо богаче, но почему-то Гитхаб более популярен.

Для облегчения разборок с git рекомендую поставить SmartGit от Syntevo. Это графическая надстройка, она бесплатна для некоммерческого использования и возьмет на себя львиную долю задач, просто превратив их в клики мышкой. С Гитхабом тоже интегрируется и может показывать в своём окне пулреквесты и комментарии.

linotex, 2021-03-17:
Также возможно есть смысл структурировать библиотеки по аналогии со стандартными. То есть весь код переместить в папку src (для каждой либы конечно). К примеру cmake ориентируется на такую стуркутру.
Это только предложение без каких-либо претензий.

AlexGyver, 2021-03-17:
> Также возможно есть смысл структурировать библиотеки по аналогии со стандартными. 
Не все стандартные библиотеки имеют папку src, да и совместимость копипасто-заменой сильно нарушится, это страшный сон новичка когда обновил либу и она сломалась)

Piranit, 2021-04-24:
поддерживаю @anklimov

в GitHub есть
- Issues
- Pull requests
- Actions
- Projects
- Wiki
- Security
- Insights
- Releases
- Tags
- Readme

при нынешней структуре получается что это сейчас общее на все библиотеки **все совместно** и все превращается в кашу

было бы очень удобно иметь на каждую отдельную библиотеку **отдельные**  Issues / Pull requests / Releases / Readme

собственно зашел сюда чтобы почитать Closed Issues по библиотеке чтобы посмотреть возможно у кого-то были такие же проблемы, но так как Issues один для всех по моему запросу нашло кучу вещей, которые вообще не к этой библиотеке (

AlexGyver, 2021-04-24:
Я использую Гитхаб как облачное хранилище, а не как инструмент для разработки. Возможно когда нибудь, когда будет время, все библиотеки переедут на отдельный аккаунт, у всех будет своя репа, релизы и всё подключим на ардуино менеджер библиотек. Но только не сегодня

rglory, 2021-05-13:
Присоединяюсь к автору, дополнительная проблема - библиотеками тяжело, если возможно вообще, пользоваться из platformio.

AlexGyver, 2021-05-16:
> Присоединяюсь к автору, дополнительная проблема - библиотеками тяжело, если возможно вообще, пользоваться из platformio.

Повторюсь, появится время - оформлю все библиотеки на отдельный аккаунт, в отдельные репы

AlexGyver, 2021-05-25:
Потратил несколько дней и всё оформил на отдельном аккаунте! Наслаждайтесь =) Этот репозиторий будет архивирован и обновляться не будет
https://github.com/GyverLibs

wl2776, 2021-05-25:
👍

linotex, 2021-05-25:
оуєєєєєє

ShohninDmitriy, 2021-05-27:
Звиздец теперь вместо одной репозитории 60 скачивать

rglory, 2021-05-27:
> Звиздец теперь вместо одной репозитории 60 скачивать

Ага, а если я захочу одну библиотеку версии 1.0 а другую 1.3 как это в одной репозитории делать?

AlexGyver, 2021-05-27:
> Звиздец теперь вместо одной репозитории 60 скачивать

А зачем тебе 60 библиотек? Качай тогда сразу все существующие для ардуино, их там 576 штук в менеджере библиотек

wl2776, 2021-05-28:
> Звиздец теперь вместо одной репозитории 60 скачивать

Вместо одно*ГО* репозитори*Я*.

Это слово мужского рода: https://ru.wiktionary.org/wiki/репозиторий