AlexGyver/GyverLibs #73: Плохая совместимость библиотек microLED и gyverButton

artistru, 2021-02-28:
При совместном использовании ломается логика выходов и входов. Для ws2812 на пине дата происходит подтяжка к земле, что приводит к течению тока с пина vcc на пин дата, можно отключить землю и все равно светодиоды будут светить. Происходит это только при вызове метода show(). В свою очередь появляются ложные срабатывания на кнопках, которые подключены к земле и подтянуты к Vcc. Если функционал microLED заменить на fastLED, то проблемы с кнопками нет. Но microLED намного практичнее и удобнее fastLED, и очень хорошо документирована. Есть ещё, как минимум один человек, который столкнулся с аналогичной проблемой. Очень ждём исправлений.

linotex, 2021-03-14:
У меня была похожая проблема (https://github.com/AlexGyver/GyverLibs/issues/63). Решилась добавлением резисторов к кнопкам и отключением внутреннего. Возможно поможет

artistru, 2021-03-14:
Резисторы есть. Вся проблема в нарушении инициализации ленты при использовании gyverButtons. Такое ощущение, что они используют какие то общие ресурсы одновременно. Ведь если использовать fastLED, то с тоже схемотехникой проблем нет

AlexGyver, 2021-03-15:
ПОЧИНИЛИ!!! Спасибо за репорт =)

closed 2021-03-15 by AlexGyver