AlexGyver/GyverLibs #74: Ошибки компиляции GyverStepper+Blynk на ESP8266

Greendumb, 2021-03-03:
При попытке компиляции простенького скетча для работы шаговика через blynk:
```
#define BLYNK_PRINT Serial
#include <ESP8266WiFi.h>
#include <BlynkSimpleEsp8266.h>
#include <GyverStepper.h>
GStepper<STEPPER2WIRE> stepper(800, D8, D7, D6);

char auth[] = "";
char ssid[] = "";
char pass[] = "";

void setup()
{
  Serial.begin(9600);
  Blynk.begin(auth, ssid, pass);
    // Шаговик
    // режим следования к целевой позиции
  stepper.setRunMode(FOLLOW_POS);

  // установка макс. скорости в шагах/сек
  stepper.setMaxSpeed(1000);

  // установка ускорения в шагах/сек/сек
  stepper.setAcceleration(1000);
 }

void loop()
{
  Blynk.run();
}

BLYNK_WRITE(V1)
{
   stepper.setTarget(param.asInt());
}
```
выдает ошибки
```
Arduino: 1.8.13 (Windows Store 1.8.42.0) (Windows 10), Плата:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (1M SPIFFS)"

In file included from C:\Users\Greendumb\Documents\Arduino\sketch_mar02a\sketch_mar02a.ino:4:0:

C:\Users\Greendumb\Documents\Arduino\libraries\GyverStepper/GyverStepper.h: In member function 'void GStepper<_DRV>::setMaxSpeed(float)':

C:\Users\Greendumb\Documents\Arduino\libraries\GyverStepper/GyverStepper.h:286:43: error: there are no arguments to 'max' that depend on a template parameter, so a declaration of 'max' must be available [-fpermissive]

   _maxSpeed = max(speed, MIN_STEPPER_SPEED); // 1 шаг в час минимум

                                           ^

C:\Users\Greendumb\Documents\Arduino\libraries\GyverStepper/GyverStepper.h:286:43: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)

C:\Users\Greendumb\Documents\Arduino\libraries\GyverStepper/GyverStepper.h: In member function 'void GStepper<_DRV>::setSpeed(float, bool)':

C:\Users\Greendumb\Documents\Arduino\libraries\GyverStepper/GyverStepper.h:364:37: error: there are no arguments to 'max' that depend on a template parameter, so a declaration of 'max' must be available [-fpermissive]

    int maxSpeed = max(speed1, speed2);

                                     ^

C:\Users\Greendumb\Documents\Arduino\libraries\GyverStepper/GyverStepper.h: In instantiation of 'void GStepper<_DRV>::setMaxSpeed(float) [with GS_driverType _DRV = (GS_driverType)0u]':

C:\Users\Greendumb\Documents\Arduino\sketch_mar02a\sketch_mar02a.ino:22:27:   required from here

C:\Users\Greendumb\Documents\Arduino\libraries\GyverStepper/GyverStepper.h:286:43: error: 'max' was not declared in this scope

   _maxSpeed = max(speed, MIN_STEPPER_SPEED); // 1 шаг в час минимум

                                           ^

C:\Users\Greendumb\Documents\Arduino\libraries\GyverStepper/GyverStepper.h:286:43: note: suggested alternative:

In file included from c:\users\greendumb\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\char_traits.h:39:0,

                 from c:\users\greendumb\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\string:40,

                 from c:\users\greendumb\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\stdexcept:39,

                 from c:\users\greendumb\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\array:38,

                 from c:\users\greendumb\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\tuple:39,

                 from c:\users\greendumb\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:55,

                 from C:\Users\Greendumb\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:27,

                 from C:\Users\Greendumb\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,

                 from C:\Users\Greendumb\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,

                 from C:\Users\Greendumb\Documents\Arduino\sketch_mar02a\sketch_mar02a.ino:2:

c:\users\greendumb\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:260:5: note:   'std::max'

     max(const _Tp& __a, const _Tp& __b, _Compare __comp)

     ^

exit status 1

Ошибка компиляции для платы NodeMCU 1.0 (ESP-12E Module).
```
Причем отдельно скетчи с библиотекой работают - мотор крутится (проверял на нескольких примерах). Блинк тоже отдельно работает.
В чем может быть проблема?
Заранее благодарю!

AlexGyver, 2021-03-15:
Проблема в индусах, которые писали ядро для есп

AlexGyver, 2021-03-15:
Сейчас проверил, всё компилится, у тебя видимо древняя версия ядра ESP. Я проверял на 2.7.4

closed 2021-03-15 by AlexGyver