AlexGyver/GyverLibs #76: GyverOLED -> setCursor - странное поведение

linotex, 2021-03-26:
Привет!
Странно, но почему-то setCursor устанавливает позицию символа только для **y**, а для **x** попиксельно. Так было задумано или просто копипаста?

AlexGyver, 2021-05-09:
Привет, копипаста откуда? Так было задумано, потому что олед аппаратно адресует столбики по 8 байт, что активно используется для работы без буфера. Если нужно ткнуть точно в Y - используй setCursorXY

closed 2021-05-09 by AlexGyver

AlexGyver, 2021-05-10:
Просто 5, зачем << 3?

linotex, 2021-05-10:
Если просто 5 то устанавливаеться в 5 пиксель, не в столбец ((. Возможно что то не так инициализирую. Код взял с метода setCursor

Там даже в коментарии указано пиксели по x
```
	// поставить курсор для символа 0-127, 0-8(4)
	void setCursor(int x, int y) { setCursorXY(x, y << 3); }
```

AlexGyver, 2021-05-10:
а что в твоём понимании столбец? Если столбец текста, то символ имеет ширину 6 пикселей, то есть просто умножай x на 6

linotex, 2021-05-11:
Для меня столбец 8 пикселей
Я делал меню для своего проекта и сделал курсор символом `>` и мне нужно было переместить курсор дальше и нарисовать иконку а потом и текст (да да, я знаю при печати оно автоматически двигает курсор). И посколько `setCursor` двигает курсор на 8 пикселей, то y я решил что также должно двигать на 8 пикселей по x и для меня было странно что это не так. Потому и задал вопрос :) Но тепер понял что так было задумано ))