AlexGyver/GyverLibs #84: После обновления packages esp8266 не компилируется скетч с библиотекой GyverTM1637.h

fess1100, 2021-05-22:
При компиляции стала появляться ошибка, связываю с обновлением esp8266. Как лечить не знаю, чем то библиотеке time.h перестала нравиться буква "u"

```
c:\users\oooooooo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08\xtensa-lx106-elf\include\sys\time.h: In function 'void bintime_addx(bintime*, uint64_t)':
C:\Program Files (x86)\Arduino\libraries\GyverTM1637/GyverTM1637.h:137:12: error: expected unqualified-id before numeric constant
  137 | #define _u 0x1c
      |            ^~~~
C:\Program Files (x86)\Arduino\libraries\GyverTM1637/GyverTM1637.h:137:12: error: lvalue required as left operand of assignment
  137 | #define _u 0x1c
      |            ^~~~
c:\users\oooooooo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08\xtensa-lx106-elf\include\sys\time.h: In function 'void bintime_add(bintime*, const bintime*)':
C:\Program Files (x86)\Arduino\libraries\GyverTM1637/GyverTM1637.h:137:12: error: expected unqualified-id before numeric constant
  137 | #define _u 0x1c
      |            ^~~~
C:\Program Files (x86)\Arduino\libraries\GyverTM1637/GyverTM1637.h:137:12: error: lvalue required as left operand of assignment
  137 | #define _u 0x1c
      |            ^~~~
c:\users\oooooooo\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08\xtensa-lx106-elf\include\sys\time.h: In function 'void bintime_sub(bintime*, const bintime*)':
C:\Program Files (x86)\Arduino\libraries\GyverTM1637/GyverTM1637.h:137:12: error: expected unqualified-id before numeric constant
  137 | #define _u 0x1c
      |            ^~~~
C:\Program Files (x86)\Arduino\libraries\GyverTM1637/GyverTM1637.h:137:12: error: lvalue required as left operand of assignment
  137 | #define _u 0x1c
      |            ^~~~
exit status 1
Ошибка компиляции для платы NodeMCU 1.0 (ESP-12E Module).
```

krukhlis, 2021-05-22:
Это не 
>чем то библиотеке time.h перестала нравиться буква "u"
, а вы где то заинклудили 2 места, где дефайнится с одним и тем же именем _u. Это запрещено стандартом языка, вы не можете редефайнить то, что уже задефайнено. 
Т.е. у вас не может быть `#define _u 0x1c` и ещё какой-то `#define _u ` где либо.
И у вас не может быть, например,
`#define _u 0x1c
`....
`const int _u 100;
`
ОФФТОПИК: Ну и да, вместо `#define _u 0x1c` правильно использовать `const int _u = 0x1c`

fess1100, 2021-05-23:
Собственно в моем скетче, к слову говоря, буква "u" даже не используется. Претензии как я понял возникли к самой библиотеке `https://github.com/AlexGyver/GyverLibs/tree/master/GyverTM1637` Просмотрел файл GyverTM1637.h на дублирование мест, ничего такого вроде не нашел